SpacehuhnTech/DeauthDetector #1: Combining DeauthAttack and DeauthDetector

J4ck02, 2017-04-16:
Would it be possible to put poth Codes on one ESP8266 and run them both simultaneously?
J4ck

andrewsalmon, 2017-05-01:
Of course it'd be possible. But it isn't very practical. Why would you want to do this? If you were to use the Deauth in your device, it could (unless programmed otherwise) set itself off.  

I won't lie, this idea crossed my mind as well however I don't see a scenario where it's useful, unless of course you only have one ESP8266 device and want to have both functionalities. 

I'm sure anybody can just mash these two git's together. Hell, even the Mac.cpp and Mac.h are (From what I paid attention to, and according to version 1.2) identical.  You could quite literally copy & paste and have this functional. (Although, for obvious reasons, you should just take a look at the code and edit it to fit, for good practice and all)

J4ck02, 2017-05-01:
Thanks for your reply But in my Case i have one Enclosure and i want to keep it as small as possible so innstalling 2 ESP8266 wouldn´t be that effective. Yeah of course it would trigger itself but if it´s a blinking LED who cares ? And u could still use it as Deauth Scanner.
Sry for my bad english it´s not my first language.
Have a nice Day

spacehuhn, 2017-05-04:
To run the detector it has to constantly scan and it can't do other stuff while doing this.

closed 2017-05-04 by spacehuhn

marek26340, 2017-05-04:
I think that it would have been better if you will combine these codes and we would be able to set what we want (attack or detect attack) every time we start up our module.

marek26340, 2017-05-04:
But i don't know pretty much anything about coding that you do (arduino), so, i don't know if that's even possible :D.

J4ck02, 2017-05-04:
It would be possible i guess and it´s a nice idea

spacehuhn, 2017-05-04:
Ok yeah I guess that's possible. Shouldn't be that hard, too.
Though I don't really have the need for that.

andrewsalmon, 2017-05-04:
I agree that it isn't hard to do and it's possible however I still don't think it's practical.  The space is already limited as is for how much code you can upload.  I think that it'd just be easier and simple to have them as separate git's and if someone wants to have both the functionality from them, they can either do it themselves Or you can write a tutorial on how to integrate deuath checker with deauther.   

I think some people would think it's cool but it really wouldn't have all that much use. That is why I think it's best that they're separate.

andrewsalmon, 2017-05-04:
Deauther has ~830 stars. That's just people that have a GitHub account and starred it. Not even including the others that didn't.

And the checker only has around ~30.

 If more people actually wanted this then ye I can see why you'd do it but as of right now it seems like only a couple people that would even use it.

J4ck02, 2017-05-04:
Yeah i think youre right it´s probably the best if everyone could choose wether use it or not.

spacehuhn, 2017-05-04:
You are free to make your own version featuring both projects.

J4ck02, 2017-05-04:
:DD if i just could....

marek26340, 2017-05-04:
:D just for fun... I'm going to add a battery with a 5V step-up converter on my NodeMCU v3. Because i can :D