SpacehuhnTech/DeauthDetector #2: arduino compilation

cookins, 2017-06-11:
hey there
i want to test your project and and found some problems in arduino compilation -

exit status 1
missing terminating ' character

i have 1.8.1 version and last esp8266 SDK.
im not changed anything in sketch. module is esp-12. arduino nano.
here full error - https://pastebin.com/6CjRVcP6

supersjimmie, 2017-06-12:
Compiles fine here.
According to your full error pastebin, I think you are not compiling the .ino file but you are trying to compile some html file.

cookins, 2017-06-26:
It's opened as ino file, if that what you mean... Or I misunderstood something?...

supersjimmie, 2017-07-06:
You may have a file with an .ino extension but what you show in the pastebin link is definitely not Arduino code.

supersjimmie, 2017-07-06:
Perhaps you can try to copy/paste this text into a new ino file and try again:
https://raw.githubusercontent.com/spacehuhn/DeauthDetector/master/deauth_detector/deauth_detector.ino

Bitofanoop, 2020-09-24:
hi, i'm trying to compile for an nodemcu esp8266 and compiling goes without errors but when i "uncomment" the  #
#define LED 2              /* LED pin (2=built-in LED) */   line
so i can use the onbiard led,,    compiling fails on that line with a "define does not name a type" error.
this is on my 3rd clean install of arduino and installing the esp8266 community 2.7.4.  i just cant seem to get it to go.
is this the place to drop this question 
regards and thanks in advance

Bitofanoop, 2020-09-25:
compiling worked. a lot to learn for me here.
another question. could the code be modified to send me a mail or a pushover notification trough a android app with api key on phone? please point me the way

wobegone, 2023-02-04:
Compile this file: https://github.com/SpacehuhnTech/DeauthDetector/blob/4386dca8c2c61687b1870b40b6b0c0e5ccc1378a/deauth_detector/deauth_detector.bin

With this tool: https://esp.huhn.me/

cookins, 2023-02-04:
Better late than never.