SpacehuhnTech/DeauthDetector #3: crash with Exception (3)

pubcoder, 2017-08-12:
It used to work, and then one day I powered it up and it kept resetting with cause:4 (watch dog?).
After I reflashed it, it just crashed with Exception (3) upon WiFi.disconnect() command.
Then I found this suggestion:
http://blog.flynnmetrics.com/uncategorized/esp8266-exception-3/

here is the patch:
```
@@ -46,7 +46,8 @@ void setup() {
 
   wifi_set_opmode(STATION_MODE);
   wifi_promiscuous_enable(0);
-  WiFi.disconnect();
+  WiFi.persistent(false); 
+  WiFi.disconnect(true);
   wifi_set_promiscuous_rx_cb(sniffer);
   wifi_set_channel(curChannel);
   wifi_promiscuous_enable(1);

```
I still don't understand what's going on, but it works, perhaps there is a bug in the library.
I use Arduino IDE version 1.8.2 and esp8266 library version 2.3.0.

Droid-MAX, 2018-06-30:
Maybe I have the same problem as you.

Droid-MAX, 2018-06-30:
You should submit a pull request

dontsovcmc, 2018-08-04:
I got the same issuu with release version of WiFiManager
https://github.com/tzapu/WiFiManager/issues/679

gordonf130, 2020-09-28:
> It used to work, and then one day I powered it up and it kept resetting with cause:4 (watch dog?).
> After I reflashed it, it just crashed with Exception (3) upon WiFi.disconnect() command.
> Then I [...]

where about in the code dose the patch go?

monkeyaction, 2020-10-01:
Same issue: ESP8266 compiles and loads but crashed with Exception (3), after decoding error:

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000bf64
EXCVADDR: 0x40270307

Decoding stack results
0x402258df: String::String(char const*) at C:\Users\aaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 34
0x40212820: jsonFlag(String&, char const*, bool) at C:\Users\aaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.h line 124
0x402258df: String::String(char const*) at C:\Users\aaa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 34
0x40212ab8: Settings::getJsonStr() at C:\Users\Nick\AppData\Local\Temp\arduino_build_692178\sketch\Settings.cpp line 101
0x40212f09: Settings::save(bool) at C:\Users\Nick\AppData\Local\Temp\arduino_build_692178\sketch\Settings.cpp line 193
0x4021a720: loop() at C:\Users\Nick\Documents\Arduino\esp8266_deauther/esp8266_deauther.ino line 188

I've spent 3 days on this so far :-(