SpacehuhnTech/DeauthDetector #7: The deauth_detector.bin file isn't available

ulmanov, 2017-11-05:
As I already mentioned on the title, the deauth_detector.bin file isn't available for download. So I unfortunately cannot flash my device and check the software out. It would be really nice, if you would upload the file.
Thanks!

spacehuhn, 2017-11-05:
Sorry, here it is: 
https://github.com/spacehuhn/DeauthDetector/raw/master/deauth_detector/deauth_detector.bin

zackkattackked, 2019-11-05:
Hi there, I am new to this and just starting out - thanks for the project!

I was looking for the same thing and have clicked the link but it seems to be not found and can't find it anywhere else on the website. Could you please re-link it or point it in the right direction for me? Thanks.

cibelix, 2019-11-05:
hi:

Same here, I need bin file please ...

Mahirmandala471, 2021-08-03:
Where  the file bin?

Mahirmandala471, 2021-08-03:
> Sorry, here it is:
> https://github.com/spacehuhn/DeauthDetector/raw/master/deauth_detector/deauth_detector.bin

file does not exist

wobegone, 2023-02-04:
Please @spacehuhn upload the bin

wobegone, 2023-02-04:
https://github.com/SpacehuhnTech/DeauthDetector/blob/4386dca8c2c61687b1870b40b6b0c0e5ccc1378a/deauth_detector/deauth_detector.bin