SpacehuhnTech/DeauthDetector #11: Buzzer Not work with Led

sebk95, 2019-01-18:
Buzzer not work with my Nodemcu v3.  In Buzzer version not switch on the Led on board nodemcu...please fix

spacehuhn, 2019-01-18:
Have you connected a buzzer to the NodeMCU?

sebk95, 2019-01-18:
Yes But not work  with this project...... other project work perfectly. For example with this work https://gist.github.com/tagliati/1804108

spacehuhn, 2019-01-18:
OK have you changed the pin in the sketch?
https://github.com/spacehuhn/DeauthDetector/blob/master/deauth_detector_buzzer/deauth_detector_buzzer.ino#L14

I'm asking because the example you sent is using pin 11 and the sketch here is using pin 5 by default.

sebk95, 2019-01-18:
yes in D4 but as soon as i connect the buzzer it always rings without recognizing the attack and the LED does not light up

sebk95, 2019-01-18:
buzzer rings with default tone (classic beep)