SpacehuhnTech/DeauthDetector #19: looking for wiring diagram you used for esp8266 & usb-C power supply

nylar357, 2021-11-30:
if you could send me any wiring diagrams for the esp8266 module and usb-c power supply id really appreciate it....bryce@motoko00.anonaddy.com

wobegone, 2023-02-04:
he doesn´t wired anything 
just: compile this file: https://github.com/SpacehuhnTech/DeauthDetector/blob/4386dca8c2c61687b1870b40b6b0c0e5ccc1378a/deauth_detector/deauth_detector.bin

With this tool: https://esp.huhn.me/

Mraanderson, 2023-12-17:
USB 3.1 type c male/Female Connectors Jack Tail 24pin usb Male Plug Electric Terminals welding DIY data cable Support PCB Board
https://a.aliexpress.com/_mrZRoqM using just VIN  and Gnd. Looks like a 3.3v Buck converter too. MP1584EN DC-DC Mini Step Down Buck Boost Converters Power Supply Module DC 3.3V 4V 5V
https://a.aliexpress.com/_mMKwTmC

USB-C to VIN on the Buck and from the 3.3v Buck to the ESP. GND to GND to GND. 

This is speculation. If in doubt don't try and just but a build from Malduino but use a USB-C to Micro adapter.