SpacehuhnTech/DeauthDetector #20: add MQTT publish

achill-es, 2022-04-16:
Wouldn't it be a nice improvement to add a function that publishes the channel number where an deauth attack has been identified (LED on) to a pre-configured MQTT broker?

The broker would implicitly record the UTC for the event for later deep analysis.

More ideas on this proposed enhancement welcome...

vongyver, 2022-07-07:
It should be possible, in the event of an attack being detected to step out of the main loop, connect to  wifi network, then mqtt server and post a record.  
Alternatively, if you have a running Node-red instance with a serial interface, you could read from the serial port and post to MQTT within Node-red, which is what I am currently doing.  I skipped the timestamp from the esp32 because modding time is a pain. 
    if (packet_rate>0) {
      Serial.print("{Channel:");
      Serial.print(channels[ch_index]);
      Serial.print(",Packets:");
      Serial.print(packet_rate);
      Serial.println("}");
    }

chris-nite, 2023-07-03:
> It should be possible, in the event of an attack being detected to step out of the main loop, connect to wifi network, then mqtt server and post a record. Alternatively, if you have a running [...]

Hi @vongyver , i'm currently trying to implement the deauth detector into my home assistant + node-red build as sensor, so i can properly alert via App, Telegram, Lights, etc.
Would you be kind enough to share your Node-Red-Template for properly splitting up the serial input into a MQTT-Topic?
Thanks in advance and best regards.

vongyver, 2023-07-03:
Sadly, I deleted the code for this.  If I recall I just was counting the
triggers and alerting when it increased within Node-Red.
Best of luck with the project! Thank you for the work.